smarmengol/Modbus-Master-Slave-for-Arduino #2: Known issues

smarmengol, 2015-03-21:
KNOWN ISSUES

It is not compatible with ARDUINO LEONARDO and not tested under ARDUINO DUE and newer boards. 
ARDUINO LEONARDO Boards aren't compatible with the add-on required for RS485 compatibility.

TODO List

Common to Master and Slave:

1) Implement other Serial settings: parity, stop bits, ...

2) End frame delay, also known as T35

3) Test it with several Arduino boards: UNO, Mega, etc..

4) Extend it to Leonardo

Master:

1) Function code 1 and 2 still not implemented

2) Function code 15 still not implement

3) Other codes under development

bilbolodz, 2017-02-21:
Any plans for Leonardo compatibility?

machoyouk, 2017-07-27:
Any plans for Leonardo compatibility?

smarmengol, 2017-08-10:
I'm not working on it at the moment.

Cloolalang, 2020-02-14:
Trying to use this with HC12 UART radios, and Arduino UNO. All is well until I tried to use it at lower baud rates (<4800) (to increase radio range). 
I tried changing some of the timer settings but no luck.
I know that radios are working well at 1200bd and have tested using other modbus Arduino libs and a windows modbus simulator.
I really like this library, it would be great to be able to use it at the lower baud rates for our radio based Modbus.
Cheers
Patrick

Gio-dot, 2020-06-26:
> Trying to use this with HC12 UART radios, and Arduino UNO. All is well until I tried to use it at lower baud rates (<4800) (to increase radio range).
> I tried changing some of the timer settings but [...]

Hi, which is modbus library that works with HC 12 at lower baud rates? I'm looking too at a similar project...

Cloolalang, 2020-06-26:
https://github.com/norgor/ModbusRTU

Cloolalang, 2020-06-26:
The Ebyte Lora Uart Radio are far superior, also checkout Mysensors,

Gio-dot, 2020-06-26:
> The Ebyte Lora Uart Radio are far superior, also checkout Mysensors,

Thank you! Do you know (using smarmengol library) how can be used modbus registers starting i.e. from 300? I'm using this library (slave mode) with modbus registers from 0 to 70 (holding registers), but i have no idea of how to use ie registers from 300 to 370 instead...