smarmengol/Modbus-Master-Slave-for-Arduino #3: CRC error when more than 29 registers

beer4duke, 2015-09-24:
Hello Samuel,
when I am trying your example RS485_slave increasing registers length to 32, it is OK to ask for 29 registers, but when hitting 30, my modbus master complains about the CRC:
>>> test = instrument.read_registers(0, 30)

MinimalModbus debug mode. Writing to instrument (expecting 65 bytes back): '\x01\x03\x00\x00\x00\x1e\xc5\xc2' (01 03 00 00 00 1E C5 C2)
MinimalModbus debug mode. No sleep required before write. Time since previous read: 3182.5 ms, minimum silent period: 1.00 ms.
MinimalModbus debug mode. Response from instrument: '\x01\x03<\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00zA' (01 03 3C 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 7A 41) (65 bytes), roundtrip time: 30.3 ms. Timeout setting: 50.0 ms.

Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/usr/lib/python2.7/site-packages/minimalmodbus.py", line 498, in read_registers
    numberOfRegisters=numberOfRegisters, payloadformat='registers')
  File "/usr/lib/python2.7/site-packages/minimalmodbus.py", line 697, in _genericCommand
    payloadFromSlave = self._performCommand(functioncode, payloadToSlave)
  File "/usr/lib/python2.7/site-packages/minimalmodbus.py", line 798, in _performCommand
    payloadFromSlave = _extractPayload(response, self.address, self.mode, functioncode)
  File "/usr/lib/python2.7/site-packages/minimalmodbus.py", line 1075, in _extractPayload
    raise ValueError(text)
ValueError: Checksum error in rtu mode: 'zA' instead of 'z\xd7' . The response is: '\x01\x03<\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00zA' (plain response: '\x01\x03<\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00zA')

As you see I am using minimalmodbus library.
Have you already seen this kind of problem?

Cheers,
    Beer4duke

smarmengol, 2016-02-27:
Thanks! I haven't seen it before.

Vincentxiaojie, 2017-01-19:
@smarmengol Any plan to fix it? Thanks.

smarmengol, 2017-01-20:
@Vincentxiaojie Not too much time for these things, but you can fix it with a new CRC calculator. Thanks!

omeranar1, 2017-06-19:
@beer4duke hi im have same problem did you fix
Thank you

letrich, 2017-07-29:
This because of small buffer size:

#define  MAX_BUFFER  64	//!< maximum size for the communication buffer in bytes

Make it larger (128 for example) and all be good.

Devangsavla, 2017-12-05:
@letrich Where do if find this line in the code?
#define MAX_BUFFER 64

letrich, 2017-12-05:
@Devangsavla 
Find the string in ModbusRtu.h

closed 2020-01-17 by smarmengol