smarmengol/Modbus-Master-Slave-for-Arduino #5: MAX_BUFFER question

0jinjing, 2016-05-09:
Hi,
I have a question with MAX_BUFFER. Default MAX_BUFFER is 64 bytes, but I'm add this value to 512. It's goes well, and give right response. However, MAX_BUFFER more than 512. My modbus Master get nothing. Is there some limit?  I'm tried to change T35's value but it's not work.

Could you illustrate it ? Thanks.

smarmengol, 2016-05-09:
Hi,
I haven't tried it. Anyway the MODBUS standard doesn't allow more than 253 bytes: 

http://www.modbus.org/docs/Modbus_Application_Protocol_V1_1b3.pdf

According to this web 

http://www.hobbytronics.co.uk/arduino-serial-buffer-size

you could. Maybe the problem is inside my code. 

Regards,

0jinjing, 2016-05-11:
Hi,
I'm trying to got au16data that au16data's size is 1024bytes and modbus master can polling 10 times to get all data, eg. first time is get 0x0~0x78, second time is 0x79~0xF0.....

"http://www.hobbytronics.co.uk/arduino-serial-buffer-size"=> Increasing the buffer size in HardwareSerial.cpp is not help save this problem.

I think it's because RAM is not enough to save "au16data" that size more than 256.
I check RAM with this and print au16data's data:
int free_ram()
{
  extern int __heap_start, *__brkval;
  int v;
  return (int) &v - (__brkval == 0 ? (int) &__heap_start : (int) __brkval);
}
It shows au16data[2xx] is not be "0", maybe it means memory conflict.

shivalingaraju, 2017-05-11:
hi Ojinjing,
I am trying to read 200 bytes of data from the RS485 is it possible. if yes how can I achieve it?

closed 2020-01-17 by smarmengol