smarmengol/Modbus-Master-Slave-for-Arduino #7: Is there an example of how to specify the address of a register?

rkertesz, 2016-07-08:
The registers in the examples start at address 0. Is there a recommended method to start at 1000, for example?

smarmengol, 2016-07-28:
It depends on the Modbus function. For the Master, there is no problem. The thing is a little more complex for the slave, which should require an offset address.

shivalingaraju, 2017-04-25:
In the Arduino code, you can change it to 1000 instead of 0 your work will be done.

alextingle, 2019-06-05:
I'm working on a shim that should add address mapping to the slave, without adding too much extra to the memory needs.

closed 2020-01-17 by smarmengol

Gio-dot, 2020-06-26:
> It depends on the Modbus function. For the Master, there is no problem. The thing is a little more complex for the slave, which should require an offset address.

Hi, how can be added this offset address?