smarmengol/Modbus-Master-Slave-for-Arduino #8: version compatibility regarding serial.flush()

Helium6072, 2016-07-28:
I wanted to use software serial instead of hardware serial because I wanted to send data to my computer via hardware serial, so I studied your library. I discovered that you used serial.flush() in your library(ModbusRtu.h line 287, 337, & 827: port->flush();), and Arduino.cc says that flush() **"Waits for the transmission of outgoing serial data to complete. (Prior to Arduino 1.0, this instead removed any buffered incoming serial data.)"** 
So your library does not work on newest versions like 1.6, right? Are there any other problem that makes this library incompatible with arduino 1.6? Hoping that you can fix this so that I can work on 1.6 version. Thx!
Ps. If you make it work with software serial, I'd be really happy :D

Helium6072, 2016-07-28:
I guess use "while(port->read() >= 0);" instead of "port->flush();" will work on 1.6. I'll test it out tomorrow and see if this works :P .

smarmengol, 2016-07-28:
I'm afraid that I haven't any free time to solve such issues. It should be great to make this library work with software serial. 

I thought it worked fine with IDE v1.6.

Helium6072, 2016-07-29:
Thanks for your answer!
It's just that flush() no longer empties incoming buffer. I think it will be better if use while(port->read() >= 0); instead so it will not run into weird issues. Since I want to use modbus on software serial, I will fork this repository and try to make it work on software serial.

Helium6072, 2016-07-29:
I solved those problems mentioned  above. I've made a pull request. Plz check out the new features.

closed 2016-07-30 by Helium6072