smarmengol/Modbus-Master-Slave-for-Arduino #11: TX pin 0 can be even for RS485

Krosik, 2016-08-06:
Hello

just small reminder to you, there can be RS485 without needing to set TX pin. I just read comments in your source code, where you presume if TX=0 it must be RS232. There are PCBs with auto setting TX pin, search on eBay "SCM TTL to RS485 Adapter".

good luck

smarmengol, 2016-08-07:
Hi,

Thanks for the information.
This function is useful when using a MAX483 and any of its derivates, where
the Arduino must command the flow control.

Best Regards

Krosik, 2016-08-08:
forgot to ask, in this code
```
    // set RS485 transceiver to transmit mode
    if (u8txenpin > 1)
    {
        switch( u8serno )
        {
#if defined(UBRR1H)
        case 1:
            UCSR1A=UCSR1A |(1 << TXC1);
            break;
#endif
```
is this code ok? because it won't run when I set txenpin 0 (for auto TX signal RS485 boards), but still it is RS485 protocol..

smarmengol, 2017-01-20:
Pin 0 is reserved for UART. Therefore it should be avoided.

alextingle, 2019-06-05:
Personally, I don't see why pins 0 or 1 cannot be used for Tx enable. If we are using software serial, or Serial1-3 on the Mega, then pins 0 & 1 are available for other uses. Why not allow them to be used for this?

closed 2020-01-17 by smarmengol