smarmengol/Modbus-Master-Slave-for-Arduino #12: Modbus::poll() will always timeout

Krosik, 2016-08-08:
Hi,

I'm not sure if I'm using the poll fuction correctly, my idea is:
```
// modbus master
void loop () {
  modbus.poll();
  if (something && modbus.getState() == COM_IDLE) {
    modbus.query(modbusCommand);
  }
}
```

problem here is, even when sending packet was done succesfully, thanks to ::poll() function, it will always do 2 wrong things:
1. increase errors counter after timeout's time thanks to this code in ::poll() function
2. it will set LastError to NO_REPLY even if packet was send successfully
```
    if (millis() > u32timeOut)
    {
        u8state = COM_IDLE;
        u8lastError = NO_REPLY;
        u16errCnt++;
        return 0;
    }
```
so as I understand it, it should be something like
`if (weStillWaitingForAnswer && millis() > u32timeOut)`

I'm correct?
thanks

Krosik, 2016-08-10:
Ok I got your idea, the ::poll() is only to check after sending of master packet to determine if packet was send. It will be nice if we can use ::poll() anytime, not only after packet was send, I think the only "IF" from my previous post need to be altered.

smarmengol, 2016-08-11:
Modbus::poll() handles the reply from the slave. As you guessed, it
requires a Modbus::query() before.

Regards

SAMUEL MARCO i ARMENGOL

Nico2018, 2018-10-26:
Hi i got the same problem but im making query before poll and i still getting the error's counter increasing and last error to NO_REPLY.
also i check that i get more sended packages than recived packages
Here's my test:
`void loop() {
  switch ( u8state ) {
    case 0:
      if (millis() > u32wait) u8state++; // wait state
      break;
    case 1:
      telegram.u8id = 1; // slave address
      telegram.u8fct = 3; // function code (this one is registers read)
      telegram.u16RegAdd = 0; // start address in slave
      telegram.u16CoilsNo = 6; // number of elements (coils or registers) to read
      telegram.au16reg = au16data; // pointer to a memory array in the Arduino
      Console.print("Query");
      Console.println(master.query( telegram )); // send query (only once)

      u8state++;
      break;
    case 2:
    
      Console.print("POLL:");
      Console.println(master.poll());
    
      if (master.getState() == COM_IDLE) {
        u8state = 0;
        u32wait = millis() + 2000;
      }
      break;
  }
  Console.print(master.getOutCnt());
  Console.print(", ");
  Console.print(master.getInCnt());
  Console.print(", ");
  Console.print(master.getErrCnt());
  Console.print(", ");
  Console.println(master.getLastError());
  for (int i = 0; i < 4; i++) {
    Console.print(au16data[i]);
    Console.print(", ");
  }
  Console.println();
}`

closed 2020-01-17 by smarmengol

rsc9421, 2022-01-11:
could someone solve it?