smarmengol/Modbus-Master-Slave-for-Arduino #15: Read/Write Boolean

dixi83, 2016-08-20:
Hello I just managed to build a simple Modbus communication between a master (Raspberry) and a Slave a Arduino used your Library. Now I want to read and write a boolean (or in Modbus terms a "coil") value from my slave which is using your library. Is this possible? And how?

Kind regards,
Martijn

smarmengol, 2016-08-22:
Hello dixi83,
The Modbus slave object allows to read and to write coils, as you ask. Indeed it allows Modbus Functions 1, 2, 3, 4, 5, 6, 15 and 16. 

Here, there is an explanation for this and some simulators to debug your code prior to connect it to the Rpi:

http://arduino-experience.blogspot.com.es/2014/02/modbus-slave-example.html

Best Regards,

dixi83, 2016-08-29:
Hello Smamengol,
Thank you for your answer! I saw your library supports functions 1 (reading) and 5 (writing single coils). I used the code from the example out of the link you gave me. But I can read and write a coil, lets put is this way: I'm not getting errors from QModBus.

[image]

But where are these "coils" located in the register or memory. Is it a variable which is written in your library?  for example I have a button(pin 8) and a 2 led's (pin 9+10) connected to a Arduino Uno. 

`#include <ModbusRtu.h>
Modbus slave(1,0,12); // I use a MAX 485 on pin 12

boolean button = 8;
boolean led1   = 9;
boolean led2   = 10;

uint16_t au16data[16];

void setup() {
  slave.begin( 19200 );
}

void loop() {
  slave.poll( au16data, 16 );

  bitWrite( au16data[0], button, digitalRead( button )); // from "advanced_slave.ino" example does not work

  digitalWrite( led1 , bitRead( au16data[1], led1 )); // from "advanced_slave.ino" example does not work

  if (au16data[4] >= 1) {
    digitalWrite(led2 , HIGH); // this works but i have to send a smallint value higher or equal to 1 to addres 4
  } else {
    digitalWrite(led2 , LOW); 
  }  
} `

Can you help me out? 

Best regards, Martijn

dixi83, 2016-08-31:
I did some more research to the modbus protocol and comparing it with the results I get from QModBus it seems to handle all the messages correct.

Answering my own question:

I explored your code I figured out you write all the received data back to the "au16data" array using the function _slave.poll( **au16data**, 16 );_  

So the received data must be somewhere inside this array. I'm looking but I have limited sources to monitor/debug these things, at the moment.

Thanks for the help till now.

At this moment I'm waiting for my Arduino Mega, were I have more serial ports so I can use the SerialMonitor and the modbus at the same time. As soon I have some more results from my monitoring/debugging.

smarmengol, 2016-08-31:
It sounds good. Good luck!

closed 2016-08-31 by smarmengol

dixi83, 2016-09-09:
I would like to confirm it all works now.  Sorry to bother you but I had not enough resources to debug serial connections. The code below works on a Mega 2560 and also on a UNO. Hope it helps others. 

```
#include <ModbusRtu.h>
Modbus slave(1,2,12); // Slave ID 1, Serial 2 of the MEGA, use a MAX 485 on pin 12.

int led1   = 9;
int led2   = 6;

uint16_t au16data[16];

unsigned long sendTimer = 0; // used for debuging

void setup() {
  Serial.begin(250000);
  slave.begin( 19200 );
  sendTimer = millis();
  pinMode(led1,OUTPUT);
  pinMode(led2,OUTPUT);
}

void loop() {
  slave.poll( au16data, 16 ); // updating the modbus data

  if (bitRead(au16data[0],led1)){ // detecting if single bit is 1 or 0
    digitalWrite( led1 , HIGH);
  } else { 
    digitalWrite( led1 , LOW);
  }
  
  analogWrite(led2, au16data[4]); // testing analog output signal

  if ((millis() - sendTimer) >= 1000) { // debug all the bit's in au16data[0]
    sendTimer = millis();
    int2bit(au16data[0]);
  }  
} 

void int2bit(uint16_t toConvert){
  Serial.print("Bits: ");
  Serial.print(bitRead(toConvert,0));
  Serial.print(bitRead(toConvert,1));
  Serial.print(bitRead(toConvert,2));
  Serial.print(bitRead(toConvert,3));
  Serial.print(bitRead(toConvert,4));
  Serial.print(bitRead(toConvert,5));
  Serial.print(bitRead(toConvert,6));
  Serial.print(bitRead(toConvert,7));
  Serial.print(bitRead(toConvert,8));
  Serial.print(bitRead(toConvert,9));
  Serial.print(bitRead(toConvert,10));
  Serial.print(bitRead(toConvert,11));
  Serial.print(bitRead(toConvert,12));
  Serial.print(bitRead(toConvert,13));
  Serial.print(bitRead(toConvert,14));
  Serial.println(bitRead(toConvert,15));
}
```
Thanks again.