smarmengol/Modbus-Master-Slave-for-Arduino #18: use RS485 with SoftwareSerial

davegroff, 2017-01-20:
I have an application where I have to use Software serial and I have an RS485 network.  When using this library and selection software serial, it automatically disables the txenpin:
void Modbus::init(uint8_t u8id)
{
    this->u8id = u8id;
    this->u8serno = 4;
    this->u8txenpin = 0;
    this->u16timeOut = 1000;
}

If I bypass this and set it to a different Pin, The program gets hung up somewhere.

Maybe a new update could address this.

smarmengol, 2017-02-05:
The original Modbus library didn't support software serial. I haven't
played with it for a long time.

TXEN should require to keep the RS485 transceiver in transmission mode
as long as bits a being sent. The modbus library just builds the
messages to be sent or received throught the UART or the serial UART,
but it doesn't control it directly.

El dg 05 de 02 de 2017 a les 12:09 -0800, en/na dsjstc va escriure:
> I also could not figure out how to use TXEN with software serial (for
> a max485).

dsjstc, 2017-02-05:
I agree, but there doesn't seem to be any way to set it when using the 
new single-arg constructor.  Would you mind adding Helium6072 to this 
issue?

It would be very helpful if example "software_serial_simple_master" 
addressed the use of TXEN, even if only to clarify that it's not 
possible -- it's fairly essential for use with a max485 chip.

> The original Modbus library didn't support software serial. I haven't
> played with it for a long time.

MathiasMoog, 2019-05-08:
The Issue is quite old, but today I had the same problem.  I solved it in https://github.com/MathiasMoog/Modbus-Master-Slave-for-Arduino/tree/fix_software_serial and I create a pull request some minutes ago. 

Nevertheless you can use the original library from smarmengol, but instead of the proposed constructor Modbus(uint8_t u8id) for SofwareSerial you call Modbus(uint8_t u8id, uint8_t u8serno, uint8_t u8txenpin) with u8serno=4.  Have a look ad th code, the 4 stands for SoftwareSerial in the implementation. Afterwards you call the void begin(SoftwareSerial *sPort, long u32speed) Method.

It works fine for me with two and three Arduino Uno and MAX 485  Drivers.

closed 2020-01-17 by smarmengol