smarmengol/Modbus-Master-Slave-for-Arduino #21: Freeze after 50 day millis rollover

garageeks, 2017-04-05:
Hi everybody,
This is the only Modbus library I found working without a hitch.

However in my 24/7 installation it is freezing every 50 days, already occurred twice with the same timeframe. I believe millis() is being used incorrectly and stops working after the 50 day rollover.
https://www.arduino.cc/en/reference/millis

I saw the code and there are a lot of direct references to millis, which is suggested to be avoided, and no time comparison such as if(currentTime - previousTime > interval) which would work fine even in the middle of a millis rollover

Any advice?

smarmengol, 2017-04-30:
Hi,

Thanks for your comments and for your feedback.

I shall implement it as you suggested as soon as I can. Indeed my
applications use to be as yours (non-stop) and I have no news regarding to
this issue.

The only explanation would be that I use old Arduino IDEs and there may be
something else.

Regards,

SAMUEL

https://www.arduino.cc/en/reference/millis

garageeks, 2017-05-01:
Hi Samuel,
Thank you for your reply. 

It seems I managed to fix it. But have no time to fork and pull request right now. You can just put our name in contribution header.

**Here is the full code**

`/**
 * @file 	ModbusRtu.h
 * @version     1.21
 * @date        2016.02.21
 * @author 	Samuel Marco i Armengol
 * @contact     sammarcoarmengol@gmail.com
 * @contribution Helium6072
 *
 * @description
 *  Arduino library for communicating with Modbus devices
 *  over RS232/USB/485 via RTU protocol.
 *
 *  Further information:
 *  http://modbus.org/
 *  http://modbus.org/docs/Modbus_over_serial_line_V1_02.pdf
 *
 * @license
 *  This library is free software; you can redistribute it and/or
 *  modify it under the terms of the GNU Lesser General Public
 *  License as published by the Free Software Foundation; version
 *  2.1 of the License.
 *
 *  This library is distributed in the hope that it will be useful,
 *  but WITHOUT ANY WARRANTY; without even the implied warranty of
 *  MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.  See the GNU
 *  Lesser General Public License for more details.
 *
 *  You should have received a copy of the GNU Lesser General Public
 *  License along with this library; if not, write to the Free Software
 *  Foundation, Inc., 51 Franklin St, Fifth Floor, Boston, MA  02110-1301  USA
 *
 * @defgroup setup Modbus Object Instantiation/Initialization
 * @defgroup loop Modbus Object Management
 * @defgroup buffer Modbus Buffer Management
 * @defgroup discrete Modbus Function Codes for Discrete Coils/Inputs
 * @defgroup register Modbus Function Codes for Holding/Input Registers
 *
 */

#include <inttypes.h>
#include "Arduino.h"
#include "Print.h"
#include <SoftwareSerial.h>

/**
 * @struct modbus_t
 * @brief
 * Master query structure:
 * This includes all the necessary fields to make the Master generate a Modbus query.
 * A Master may keep several of these structures and send them cyclically or
 * use them according to program needs.
 */
typedef struct
{
    uint8_t u8id;          /*!< Slave address between 1 and 247. 0 means broadcast */
    uint8_t u8fct;         /*!< Function code: 1, 2, 3, 4, 5, 6, 15 or 16 */
    uint16_t u16RegAdd;    /*!< Address of the first register to access at slave/s */
    uint16_t u16CoilsNo;   /*!< Number of coils or registers to access */
    uint16_t *au16reg;     /*!< Pointer to memory image in master */
}
modbus_t;

enum
[... 1406 lines cut ...]
void loop() {
  switch( u8state ) {
  case 0: 
    if (millis() - u32wait > u32interval) u8state++; // wait state
    break;
  case 1: 
    telegram.u8id = 1; // slave address
    telegram.u8fct = 3; // function code (this one is registers read)
    telegram.u16RegAdd = 1; // start address in slave
    telegram.u16CoilsNo = 4; // number of elements (coils or registers) to read
    telegram.au16reg = au16data; // pointer to a memory array in the Arduino

    master.query( telegram ); // send query (only once)
    u8state++;
    break;
  case 2:
    master.poll(); // check incoming messages
    if (master.getState() == COM_IDLE) {
      u8state = 0;
      u32wait = millis();
      u32interval = 100;
    }
    break;
  }
}`

shivalingaraju, 2017-05-04:
garageeks i want to know how you are fetching data. I am using some 3rd party program to ensure I am getting the correct query from my Arduino. Communication is happening perfectly between the Arduino and the 3rd party software I know which data I am sending to the 3rd party software but in Arduino, I am not able to get that data I am using " software_serial_simple_master " example code. Awaiting for your reply thank you.

ogurets, 2018-07-13:
@garageeks why wouldn't you pastebin your code here? I've got time to do a pull request.

garageeks, 2018-07-13:
@ogurets as I said more than one year ago, GitHub doesn't allow to pastebin the entire code here. I sent the code to the author, which didn't make a commit yet.

Anyway, the fix involves replacing the millis code to avoid rollover like this:

PREVIOUS
if (millis() > u32timeOut)
FIXED
if (millis() - u32timeOut > (unsigned long) u16timeOut)

ogurets, 2018-07-13:
Hmm, let me try: https://pastebin.com/7FEi0FFC

Anyways, I've noticed a similar fix is already requested for pulling here: https://github.com/smarmengol/Modbus-Master-Slave-for-Arduino/pull/17/commits/5e0df95f0938f429ea23226a7f005dbe34d41d1c
So there doesn't seem to be a need to create another.

ueberBrot, 2018-11-06:
Did anyone tried it out after the commits from @gabrielsan ? Does t still freeze?

gabrielsan, 2018-11-06:
@ueberBrot  I have them up-and-running for about a year now, no issues so far.

Overflow is not a problem in this case, https://stackoverflow.com/questions/7221409/is-unsigned-integer-subtraction-defined-behavior

ueberBrot, 2018-11-06:
Awesome!

closed 2020-01-17 by smarmengol