smarmengol/Modbus-Master-Slave-for-Arduino #23: Not Able to See the data on the Serial monitor

shivalingaraju, 2017-04-25:
I am able to achieve RS485 Modbus communication. At the other end device, I am able to see proper acknowledgment.
[image]

but , I am not able to see any data on Arduino serial comport please do help me.  

[image]

geoffmiles, 2017-04-29:
Im having the same problem. To my knowledge the response from the modbus slave device is processed by the function get_FC3 which moves them from the au8buffer to au16reg however I am struggling to get data from them. I used a 3rd party program to ensure I am getting the correct query from my arduino and Ive used the same program to send that query to the modbus device and received the expected response. So I know the data is there I am just having some issues displaying them.

smarmengol, 2017-04-30:
The Arduino Serial Monitor shows ASCII data. As you know, the Modbus RTU
protocol handles raw binary data. Therefore it only shows ⬜ and wrong
symbols instead of the real values.

When I wrote this library, I used third-party simulators either for the
Master or for the Slave. I recommend you these:

QModbus: http://qmodbus.sourceforge.net/

Modpoll and it's slave Variant: http://www.modbusdriver.com/modpoll.html

Multiway: https://www.support-omron.fr/Multiway.php

I hope that this would help you. You can also find further information in
my blog:

http://arduino-experience.blogspot.com.es/

Regards,

SAMUEL

shivalingaraju, 2017-05-02:
Thank you for your response, sir (smarmengol),
Qmodbus version 0.2.1 and 0.1.0 I downloaded and tried I am getting following errors.

if I click on send,
[image]

sometimes I am getting this error for version 0.2.1 and for 0.1.0 also I am getting the same error.

[image]

Version 0.2.0 also I downloaded by I am not getting how to use it, sir.

shivalingaraju, 2017-05-04:
please do update if anybody got any solution to fetch the data using RS485 communication protocol.

smarmengol, 2017-05-04:
Regarding to the RS485, look at this:

http://arduino-experience.blogspot.com.es/2016/01/rs-485-implementation-on-arduino.html?m=1

shivalingaraju, 2017-05-04:
Thank you i will look into it

shivalingaraju, 2017-05-08:
thank you for your help and support it worked.

closed 2017-05-08 by shivalingaraju