smarmengol/Modbus-Master-Slave-for-Arduino #26: Timeout error advanced master

vrusu7, 2017-08-09:
I think in your example 
case 1 
……
if (u8query > 2) u8query = 0;
must be  
if (u8query == 2) u8query = 0; or if (u8query > 1) u8query = 0;

smarmengol, 2017-08-10:
The code works as it is written. I checked it.

After sending any query, the Master waits for the slave answer. This
implies that the slave must be able to process the query, arrange an answer
and send it through the serial port.

vrusu7, 2017-08-11:
Your code work fine, I used this class in my project  to realize communication between two arduino via RS485 and monitor error state, so thank you very much for your work.
About my issue:
I use Slave Arduino Nano -> RS485 -> RS485 -> Master Arduino Uno
In master I put LCD an print count of Output, Input, Error and last Error messages.
When used 
if (u8query > 2) u8query = 0;
 code work but we have  around 30% of timeout errors, so I examined code step by step with statement of u8state, u8query and I found in this case master try send undefined   third element of array telegram[2] (array  declare only two elements telegram[0], telegram[1] ) and we have  timeout error.
I modify
 if (u8query > 2) u8query = 0; to if (u8query == 2) u8query = 0; 
and code work fine without any errors.

smarmengol, 2017-08-11:
I checked it like you with two arduinos. I don't understand what may be
wrong.

Anyway replace the "==" with an ">=". It shall be safer for your code.

closed 2017-08-11 by vrusu7