smarmengol/Modbus-Master-Slave-for-Arduino #30: RS-485 half duplex operation with MAX485 and compatible transceivers

GITChr, 2017-10-30:
Hi!
Great that someone wrote this library. A question:

In the examples, RS-485 is sometimes mentioned in the code (advanced master), while at the top of the file, it is mentioned this works only with RS232 and USB. 

For operation with RS-485, the transceiver needs to be switched back and forth between receive and transmit modes. Is this implemented in the library and how do I use this feature?

Thanks
Christoph

adkolimp, 2018-05-26:
I am not too sure if you have already found a solution. If I am right you may use the following

`Modbus(uint8_t u8id, uint8_t u8serno, uint8_t u8txenpin);` If you initialise with this method then the toggling is taken care. u8txenpin is probably what you are looking for. On the RS-485 you should short RE, DE pins.
Let me know if you have solved it in any other way.

jjassar, 2018-07-20:
@GITChr  have you solved the issue with solution recommended by @adkolimp .
Please update

smarmengol, 2018-07-20:
Sorry for the delay in my answer.
The original library handled the RS-485 transceiver through a digital output:
- This pin number is declared at u8txenpin.
- The pins DE/RE must be wired to this digital output.
- Whenever the library needs to fetch (read) from the network, the transceiver is set as read mode, this is to say, "data goes into the Arduino".
- Whenever the library needs to write to the network, the transceiver is set to transfer mode, this is to say, "Arduino sends data".
The Arduino loses the u8txenpin, because the Library is already using it.

xzsfd2457, 2020-01-17:
I am getting response from devices by using below code.......
#define DE PB9
void setup() {
disableDebugPorts();
  Serial.begin(9600);
Serial1.begin(9600);
pinMode(DE,OUTPUT);
  //digitalWrite(DE,HIGH);

}
byte test_data[] ={0x01,0x04,0x00,0x10,0x00,0x01,0x30,0x0F}  ;                                                                     

void loop() 
{
  Serial1.begin(9600);
  digitalWrite(DE,HIGH);
  delay(2000);
  Serial1.write(test_data,8);
    delay(10);
digitalWrite(DE,LOW);
Serial1.end();

Serial1.begin(9600);
 while(1){
   if (Serial1.available())
  {
    Serial.println(Serial1.read(),HEX);
  }
 }
 Serial.flush();
 Serial1.flush();
  delay(1000);
  // put your main code here, to run repeatedly:

}

smarmengol, 2020-01-17:
> I am getting response from devices by using below code.......
> #define DE PB9
> void setup() {
> disableDebugPorts();
> Serial.begin(9600);
> Serial1.begin(9600);
> pinMode(DE,OUTPUT); [...]

Hi,
There is something that I'm missing. Maybe I'm getting too old.
What is the relation between this and Modbus? Your code is for a serial transmission. Nothing else!
Regards,

closed 2020-01-17 by smarmengol

xzsfd2457, 2020-01-20:
> > I am getting response from devices by using below code.......
> > #define DE PB9
> > void setup() {
> > disableDebugPorts();
> > Serial.begin(9600);
> > Serial1.begin(9600);
> > pinMode(DE,OUTPUT);
> > //digitalWrite(DE,HIGH);
> > }
> > byte test_data[] ={0x01,0x04,0x00,0x10,0x00,0x01,0x30,0x0F} ;
> > void loop()
> > {
> > Serial1.begin(9600);
> > digitalWrite(DE,HIGH);
> > delay(2000);           //Keep delay is important
> > Serial1.write(test_data,8);
> > delay(10);
> > digitalWrite(DE,LOW);
> > Serial1.end();
> > Serial1.begin(9600);
> > while(1){
> > if (Serial1.available())
> > {
> > Serial.println(Serial1.read(),HEX);
> > }
> > }
> > Serial.flush();
> > Serial1.flush();
> > delay(1000);
> > // put your main code here, to run repeatedly:
> > }
> 
> Hi,
> There is something that I'm missing. Maybe I'm getting too old.
> What is the relation between this and Modbus? Your code is for a serial transmission. Nothing else!
> Regards,

I am sending serial data using MAX485 converter...