smarmengol/Modbus-Master-Slave-for-Arduino #41: How work with two slave

svetlitsky, 2019-02-25:
Hello! I can not understand and in the examples there is no option when the master works with two or more slave devices. If there are examples - I will be grateful for the help.

smarmengol, 2019-02-25:
Hi,
As a "quick and dirty" example, I would suggest you to modify the current advanced Master and declare the second telegram for another slave:

  // telegram 0: read registers
  telegram[0].u8id = 1; // slave address
  telegram[0].u8fct = 3; // function code (this one is registers read)
  telegram[0].u16RegAdd = 0; // start address in slave
  telegram[0].u16CoilsNo = 4; // number of elements (coils or registers) to read
  telegram[0].au16reg = au16data; // pointer to a memory array in the Arduino

  // telegram 1: read a single register from another slave
  telegram[1].u8id = 2; // another slave address 
  telegram[1].u8fct = 3 // function code
  telegram[1].u16RegAdd = 0; // start address in slave
  telegram[1].u16CoilsNo = 4; // number of elements (coils or registers) to read
  telegram[1].au16reg = au16data+4; // pointer to a memory array in the Arduino

The rest of the code would be the same. Try it, because this simple works.

Cheers,

smarmengol, 2019-02-25:
If you need further information about connecting several devices through RS-485, please have a look at my blog:

https://arduino-experience.blogspot.com/

svetlitsky, 2019-02-26:
Thanks for the tip! For a complete understanding, need to specify that to exchange different arrays for different slaves, you must specify the names of the arrays with a shift in the elements of the telegram. Be sure to specify the total number of exchange elements in the line "if (u8query> 10) u8query = 0;" where 10 is the total number of elements for all slaves.

Many thanks for the library, it works fine!

Svetlitsky Sergey
svetlitsky.sv@gmail.com
+7 (912) 4872526

closed 2019-05-25 by smarmengol