smarmengol/Modbus-Master-Slave-for-Arduino #55: float type data

aasirmutwalli22, 2020-08-27:
What I have to do if float array is needed to poll

`
uint16_t modbus_array[] = {3,2,3.2};
bus.poll(modbus_array, sizeof(modbus_array) / sizeof(modbus_array[0]));
`

ulkersinan, 2021-04-02:
you can multiply 10 or 100 
Ex: dht temp value 10.25 -->> *100  = sending value on array :1025
And divide 100 after communication and convert float again while printing. -->> 1025/100 = 10.25 

PS: Sorry for my bad english.

githubShuaiGe, 2022-01-30:
Is there any other way

arthurtarasov, 2022-01-30:
@githubShuaiGe It is a bad idea to use floats with voltages and money so I don't think they have a place in this library. You can store a dividend integer in one register and a divisor integer in another (10, 100, 1000, etc.). You can retrieve both registers at the same time and convert into decimal closer to the user interface where you will have more computing power and lower cost of error.

closed 2022-02-06 by smarmengol