smarmengol/Modbus-Master-Slave-for-Arduino #57: how to send a broadcast?

gicking, 2021-02-23:
hello,

thanks a lot for this nice library! I am using it for a (very) simple Modbus master to test a RS485 network. So far all worked well. But now I want to send broadcast messages to all connected slaves, and it seems like nothing is sent (checked with oscilloscope).

Specifically in [ModbusRtu.h](https://github.com/smarmengol/Modbus-Master-Slave-for-Arduino/blob/master/ModbusRtu.h) line 55 mentions u8id=0 is intended for broadcast, but line 546 in Modbus::query() returns immediately if u8id==0.

Do I need to call a different method than query() for sending a broadcast and if yes, which one? For a simple example thanks a lot in advance! Have a great day  :-)

gicking, 2021-02-23:
FYI, as I have some time pressure I made a quick fix, which seems to work. 

change line 546 from
   if ((telegram.u8id==0) || (telegram.u8id>247)) return -3;
to
   if (telegram.u8id>247) return -3;

and change line 621from
   u8state = COM_WAITING;
to
   if (telegram.u8id!=0)
     u8state = COM_WAITING;
   else
     u8state = COM_IDLE;

imanfreeman, 2021-07-13:
I need broadcast too. However, your changes didn't work for me. 
I change the line 751 from: 
if (au8Buffer[ ID ] != u8id ) return 0;
to 
 if (au8Buffer[ ID ] != u8id   && au8Buffer[ ID ] !=0 ) return 0;
and now, the slave accepts 0 id as well.
Although, the slave processes all the funcions with this id (for example read holding registers), it returns response error not a valid one. 
Maybe there is a better and concrete solution.

linjovi168, 2021-08-13:
imanfreeman  
It is useful to change this code. 
Is there a way to cancel the  returns response in broadcast ?