smarmengol/Modbus-Master-Slave-for-Arduino #67: Isue with write data to slave

mati294, 2022-01-23:
Hello trying to send data via functions 6,15,16 to slave. But strange values ​​are blown at the slave. With the function 5 and the value 6 in the slave there are 1798 ...
Please fix it, or if I misuse it, add an example to the library on how to properly use all the functions for slave and master.
Greetings

smarmengol, 2022-02-06:
No success with the current examples?

closed 2022-02-06 by smarmengol

mati294, 2022-02-11:
It only worked after destroying all the libraries that contained something related to modbus and reinstalling the library.
However, there is still some decent example showing the relationship between functions, etc.
It would also be useful to have info on the limits of the functions (how many data cells can be sent with one telegram, etc.)